schenlap/venus_kostal_pico #2: Power values do not change when PV Inverter is disconnected

mmeintel, 2020-11-01:
HI Schenlap,

I just used your python script to get the Piko connected to a CCGX.
Thanks for that!

I found an issue that when the Piko is disconnected the load is still active and the device does not disappear from the Schematics Page in CCGX (2.60).

I am not allowed to create a PR, so I will just add the patch right here.

```bash
iff --git a/kostal.py b/kostal.py
index f19a267..bd7610e 100755
--- a/kostal.py
+++ b/kostal.py
@@ -330,6 +330,18 @@ def kostal_update_cyclic(run_event) :
                        Kostal.stats.last_connection_errors = 0
                        Kostal.stats.reconnect += 1
                        kostal.set('/Connected', 0)
+                       kostal.set('/Ac/L1/Current', None)
+                       kostal.set('/Ac/L2/Current', None)
+                       kostal.set('/Ac/L3/Current', None)
+                       kostal.set('/Ac/L1/Power', None)
+                       kostal.set('/Ac/L2/Power', None)
+                       kostal.set('/Ac/L3/Power', None)
+                       kostal.set('/Ac/L1/Voltage', None)
+                       kostal.set('/Ac/L2/Voltage', None)
+                       kostal.set('/Ac/L3/Voltage', None)
+                       kostal.set('/Ac/Power', None)
+                       kostal.set('/Ac/Current', None)
+                       kostal.set('/Ac/Voltage', None)
 
                if dev_state == DevState.WaitForDevice:
                        if kostal_read_status(init=1) == 0:
```

Best regards,
Martin

schenlap, 2020-11-01:
Hi,

Setting connected to 0 should be enough to let it disappear vom ccgx. Your changes should not harm either, so if it works for you I can add it.

PR is allowed, you first have to create a fork and push your changes.

mmeintel, 2020-11-01:
Thank you for the info, would be great if you add it.
It works well for me :)

The reason why I set the values to None is to indicate, that the data is invalid (I got this from vedbus.py).

Best regards.

schenlap, 2022-02-05:
fixed with c04c24b7

closed 2022-02-05 by schenlap