schenlap/venus_kostal_pico #4: Plenticore support needed

JostB-ger, 2022-02-04:
I've installed it on VenusOS 2.8 and got following output in the terminal:

++++++++++
POWER Phase A: 0W
POWER Phase B: 0W
POWER Phase C: 0W
POWER Total: 0W
KOSTAL Status: Parse Error
Thread: doing
a bytes-like object is required, not 'str'
parsing error, using default values for version 1 at 1
{"PT": 0, "PA": 0, "PB": 0, "PC": 0, "VA": 230, "VB": 230, "VC": 230, "EFAT": 0, "STATUS": "Parse Error", "IA": 0, "IB": 0, "IC": 0, "IN0": 0}
++++++++++

Actual optput is 277 Watts in total, not 0. So i guess the pluging is not working?
I'm using a Plenticore plus 10 not a Piko, so I would appreciate Plenticore support

schenlap, 2022-02-05:
It was introduced with the latest victron online update. Should be fixed with e29ed48dea4fdbde71dd1a547f4a426ac5b6855c

JostB-ger, 2022-03-13:
still not working on 2.82 (large 30), fix is implemented via nano editor
```
root@raspberrypi2:/data/venus_kostal_pico# python kostal.py
kostal.py:21: PyGIDeprecationWarning: GObject.idle_add is deprecated; use GLib.idle_add instead
  from gi.repository.GObject import idle_add
kostal.py:82: DeprecationWarning: The SafeConfigParser class has been renamed to ConfigParser in Python 3.2. This alias will be removed in future versions. Use ConfigParser directly instead.
  parser = SafeConfigParser()
Using config: kostal.ini
Using http://192.168.179.22 user: admin
/data/venus_kostal_pico/kostal_inverter.py starting up
Thread: doing
kostal.py:464: PyGIDeprecationWarning: Since version 3.11, calling threads_init is no longer needed. See: https://wiki.gnome.org/PyGObject/Threading
  gobject.threads_init()
kostal.py:465: PyGIDeprecationWarning: GObject.MainLoop is deprecated; use GLib.MainLoop instead
  mainloop = gobject.MainLoop()
list index out of range
parsing error, using default values for version 1 at 46
{"PT": 0, "PA": 0, "PB": 0, "PC": 0, "VA": 230, "VB": 230, "VC": 230, "EFAT": 0, "STATUS": "Parse Error", "IA": 0, "IB": 0, "IC": 0, "IN0": 0}
++++++++++
POWER Phase A: 0W
POWER Phase B: 0W
POWER Phase C: 0W
POWER Total: 0W
KOSTAL Status: Parse Error
Thread: doing
list index out of range
```

kaeferfreund, 2022-03-30:
Plenticore also not working for me :( 

I see the same error message as @JostB-ger , any advice?

kaeferfreund, 2022-03-30:
ok i see the issue - plenticore does not have a rest api afaik. I have added a pymodbus script into this repo, will submit a pr when it is tested

schenlap, 2022-04-02:
please open a PR when done.

kaeferfreund, 2022-04-08:
I tried many things, forgot half of it and tried other things.

Long story short: It seemed less initial work to use a dbus project that already used modbus and slap the plenticore part onto it. Here is the repo:

https://github.com/kaeferfreund/dbus-kostal

I'm currently thinking about slapping al the different bindings (Kostal Plenticore & Pico, Fronius, Solaredge etc.) onto one unified binding that is well tested and well tuned regarding delay and jitter.

closed 2023-03-06 by schenlap