schenlap/venus_kostal_pico #10: Kostal Piko an einem Multiplus 2 GX meldet Permission denied

Heini1985, 2023-02-19:
Hallo zusammen.
ich habe einen Multiplus GX im Einsatz.
ich habe alles nach Anleitung gemacht aber er führt den Script nicht aus (siehe Bild)
[image]

Die Ordner Struktur sieht wie folgt aus:
[image]

Remote Konsole auch nach Neustart kein kostal:
[image]

Vielleich kann mir einer helfen?

Viele Grüße

Heini1985

Heini1985, 2023-02-28:
Danke für den TIP.

ich habe es jetzt ausgeführt und bekomme das in ssh ausgegeben:

[image]

mein Remote Console sieht jetz so aus:

[image]

der PV Wechselrichter ist jetzt natürlich aus weil keine Sonne mehr vorhanden ist.
ich schaue morgen nach ob es funktioniert hat

VG Heini1985

schenlap, 2023-03-01:
Scheint so als ob das script 2 mal läuft, kann das sein?

Heini1985, 2023-03-01:
kann sein

Heini1985, 2023-03-01:
kann man das löschen und von vorne Beginen?

schenlap, 2023-03-01:
Mit ps aux /oder eventuell auch nur ps siehst du alle laufenden Prozesse. Mit kill und der Prozessnummer kannst du den beenden. Er wird irgendwas mit "python kostal.py" oder start_kostal_pico heißen. Die Prozessnummer müsste in der 2ten Zeile angezeigt werden.

Heini1985, 2023-03-01:
ich habe die Zeilen zu kostal pico aus  /data/rc.local gelöscht --> dann neu gestartet-->pico war weg.
Dann habe ich alles nochmal nach Anleitung gemach:

# Install
If setting up root access (preferable with ssh key), create a folder in data and add start script
ssh int venus
```
mkdir /data/venus_kostal_pico
echo -e '#!/bin/bash' >> /data/rc.local
echo '(cd /data/venus_kostal_pico/ && ./start_kostal_pico)' >> /data/rc.local
chmod +x /data/rc.local
exit

in ssh sah es dann so aus.

![image](https://user-images.githubusercontent.com/76739980/222221103-18dfab8a-7f8d-4c1f-af26-bb4d7ef47d29.png)

Und jetzt scheint es zu funktionieren er hat zu mindestens den Zähler ausgelesen zu habe und die Spannungen sind jetzt auf 0V weil kein Sonne mehr das ist

![image](https://user-images.githubusercontent.com/76739980/222220864-d956e51b-a37b-4039-b6e9-59833bc35788.png)

Ich schaue mal morgen od er daten liefert wenn ja dann ist es top 

Danke für die Hilfe

VG

Heini1985

Heini1985, 2023-03-01:
ich habe die Zeilen zu kostal pico aus /data/rc.local gelöscht --> dann neu gestartet-->pico war weg.
Dann habe ich alles nochmal nach Anleitung gemach:

Install
If setting up root access (preferable with ssh key), create a folder in data and add start script
ssh int venus

mkdir /data/venus_kostal_pico
echo -e '#!/bin/bash' >> /data/rc.local
echo '(cd /data/venus_kostal_pico/ && ./start_kostal_pico)' >> /data/rc.local
chmod +x /data/rc.local
exit

in ssh sah es dann so aus.

[image]

Und jetzt scheint es zu funktionieren er hat zu mindestens den Zähler ausgelesen zu habe und die Spannungen sind jetzt auf 0V weil kein Sonne mehr das ist

[image]

Ich schaue mal morgen ob er daten liefert wenn ja dann ist es top 

Danke für die Hilfe

VG

Heini1985

Heini1985, 2023-03-02:
Es hat funktioniert.

Danke für die Hilfe.

Du kannst das Thema schießen.

closed 2023-03-06 by schenlap