schenlap/venus_kostal_pico #12: PV Inverter equals Consumption

AK-868, 2023-09-12:
Hi,
i have got the problem that PV-Power is shown as consumption when charging the battery.
Doesn#t matter which mode i choose.
I have got a power-switch where in can  choose between Kostal inverter on Critical Loads or AC Loads.

When i charged my battery and discharged it.
For example tody.
I charged only the battery and had a consumption close to the PV Inverter production.

Does anyone have a idea what to do?

cheers.
AK-868

jangatzke, 2024-06-02:
I cannot reproduce this behavior, but I am seeing a higher consumption when the Pico is active. My idle consumption without sun is 300 W and about 400 W when the sun is shining. I guess the values the Pico delivers are a bit off?

xristostsilis, 2025-06-05:
I can confirm this one also on my side. PV = Consumption on VRM. It's not a big problem tho. I also cannot seem to make the inverter appear in the first display of the GX. This is ok but I would like to inform you guys.

xristostsilis, 2025-06-15:
> [@xristostsilis](https://github.com/xristostsilis) Are you using 3.60?

No. 3.55

[image]