schenlap/venus_kostal_pico #13: Display name

knirma, 2024-04-12:
Hello,

i'm using this script in Venus OS 3.30 with a Kostal Piko 5.5 NG.
Means i use version 3 in the INI file.
Until now it works.

Following question, how can i change the display name ?
in the configuration i see the value  kostal_piko_5_5
But in the front view i see only a -

Many thanks for the help

knirma, 2024-04-12:
[image]
[image]
[image]

centro-max, 2024-04-13:
> Zwischenablage01.png (view on web)
> Zwischenablage02.png (view on web)
> Zwischenablage03.png (view on web)

[KOSTAL]
ip = http://192.168.XXX.YYY
inverter_name = Carport_2_28_kWp
intervall = 10
version = 2
position = 0